JIA-Lab-research/VoxelNeXt #1: how is the performance that centerpoint upsample 32x?

AndyYuan96, 2023-03-21:
Hi, thanks for sharing the code, after reading paper, I see you use 32x feature to enlarge ERF, and centerpoint's max downsample is 16 from 2d_backbone, so I just wonder whether you have the performance that centerpoint also use max downsample 32 to make a fair compare。

yukang2017, 2023-03-21:
Hi Andy,

Thanks for your interest in our paper.

I never tried to use CenterPoint with the 32x backbone. Thanks for your remind. I will add this ablation.
In addition, I suppose the comparison is fair. The 32x backbone is to improve ERF, while the dense neck / rpn of CenterPoint is also for this purpose. Our design, "32x backbone" serves as a substitute of "8x backbone + dense neck". Anyway, I will do this experiment. I will report the result here after I finish it.

Yes, this is reasonable. VoxelNeXt is just what you said, more downsample 3D backbone + sparse head. 

Regards,
Yukang Chen

AndyYuan96, 2023-03-22:
> Hi Andy,
> 
> Thanks for your interest in our paper.
> 
> I never tried to use CenterPoint with the 32x backbone before. Thanks for your remind. I will add this ablation. In addition, I suppose the comparison is fair. The 32x backbone is to improve ERF, while the dense neck / rpn of CenterPoint is also for this purpose. Our design, "32x backbone" serves as a substitute of "8x backbone + dense neck". Anyway, I will do this experiment. I will report the result here after I finish it.
> 
> Yes, this is reasonable. VoxelNeXt is just what you said, more downsample 3D backbone + sparse head.
> 
> Regards, Yukang Chen

thanks for the reply.

closed 2023-03-22 by AndyYuan96