JIA-Lab-research/VoxelNeXt #2: Tracking on Nuscenes

Idate96, 2023-03-22:
Hi thanks for the contribution,
I was trying to find where you extract the query voxels and associate them across frames, but I can't find it in the code. Could you please point me to it?

yukang2017, 2023-03-23:
Hi,

Thanks for your interest in our work. The tracking part will be released latter. I am preparing the code. Thanks for your understanding.

Regards,
Yukang Chen

lukas-staecker, 2023-03-27:
Hi @yukang2017, do you have a rough estimate when you're going to release the tracking part?

wanghangege, 2023-04-02:
> Hi @yukang2017, do you have a rough estimate when you're going to release the tracking part?

I need your tracking test too, thanks for your work..

sayhiOK, 2023-04-13:
@yukang2017 I also need your tracking part of the code, looking forward to your sharing

QiukuZ, 2023-04-24:
I also really need the tracking part, looking forward to the early release！！

FlorianWeiser, 2023-05-05:
Hi @yukang2017, could you please release the Nuscenes tracking scripts? This would be very helpful!

Hedgehog715, 2023-05-05:
I need your tracking test too, thanks for your work.

Zzxxgg6, 2023-05-28:
Hi, @yukang2017, are the training command for the tracking task and the detection task together?

shashankyld, 2023-12-06:
is the tracking code released yet?
I need association ids of cars over multiple frames?