JIA-Lab-research/VoxelNeXt #3: test nuscenes dataset

huihui6666, 2023-03-23:
@yukang2017 
hello，I use your work to test nuscenes datasets，but the results is not very well .And the weight is not match with the model,I use the **cbgs_voxel0075_voxelnext.yaml** to test your pre-trained model trained with the nuscenes dataset . The result is show as follows

INFO  ==> Loading parameters from checkpoint /home/mia/OpenPCDet/pretrain_model/voxelnext_nuscenes_kernel1.pth to GPU
2023-03-23 16:42:13,536   INFO  Not updated weight backbone_3d.conv_input.0.weight: torch.Size([16, 3, 3, 3, 4])
2023-03-23 16:42:13,536   INFO  ==> Done (loaded 541/542)

mAP: 0.1039
mATE: 0.4398
mASE: 0.2907
mAOE: 0.7989
mAVE: 0.7291
mAAE: 0.2989
NDS: 0.2962
Eval time: 86.0s

Per-class results:
Object Class    AP      ATE     ASE     AOE     AVE     AAE
car     0.438   0.260   0.169   0.458   0.753   0.330
truck   0.117   0.450   0.221   0.415   0.641   0.318
bus     0.155   0.500   0.205   0.591   1.913   0.432
trailer 0.000   0.776   0.255   0.926   0.446   0.199
construction_vehicle    0.000   0.804   0.489   1.412   0.140   0.331
pedestrian      0.100   0.309   0.307   1.489   0.941   0.321
motorcycle      0.036   0.279   0.272   1.022   0.724   0.346
bicycle 0.000   0.299   0.293   0.786   0.273   0.113
traffic_cone    0.016   0.286   0.408   nan     nan     nan
barrier 0.177   0.434   0.288   0.089   nan     nan
2023-03-23 16:54:41,291   INFO  ----------------Nuscene detection_cvpr_2019 results-----------------
***car error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.26, 0.17, 0.46, 0.75, 0.33 | 33.53, 41.90, 47.20, 52.54 | mean AP: 0.4379142060810768
***truck error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.45, 0.22, 0.42, 0.64, 0.32 | 4.76, 9.66, 14.33, 18.14 | mean AP: 0.11723098807042298
***construction_vehicle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.80, 0.49, 1.41, 0.14, 0.33 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***bus error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.50, 0.21, 0.59, 1.91, 0.43 | 6.65, 12.97, 19.10, 23.12 | mean AP: 0.15460599829470253
***trailer error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.78, 0.26, 0.93, 0.45, 0.20 | 0.00, 0.00, 0.00, 0.04 | mean AP: 9.730399922215814e-05
***barrier error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.43, 0.29, 0.09, nan, nan | 8.99, 16.68, 21.40, 23.60 | mean AP: 0.17667373910591053
***motorcycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.28, 0.27, 1.02, 0.72, 0.35 | 2.90, 3.76, 3.84, 3.95 | mean AP: 0.03611132627896975
***bicycle error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.30, 0.29, 0.79, 0.27, 0.11 | 0.00, 0.00, 0.00, 0.00 | mean AP: 0.0
***pedestrian error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.31, 0.31, 1.49, 0.94, 0.32 | 7.60, 9.86, 10.51, 12.15 | mean AP: 0.10028934559384498
***traffic_cone error@trans, scale, orient, vel, attr | AP@0.5, 1.0, 2.0, 4.0
0.29, 0.41, nan, nan, nan | 1.03, 1.23, 1.59, 2.72 | mean AP: 0.016428388021770043
--------------average performance-------------
trans_err:       0.4398
scale_err:       0.2907
orient_err:      0.7989
vel_err:         0.7291
attr_err:        0.2989
mAP:     0.1039
NDS:     0.2962

yukang2017, 2023-03-23:
Following the previous reply, I used the following script for testing.

```
cd tools
name="cbgs_voxel0075_voxelnext"
bash scripts/dist_test.sh 4 --cfg_file cfgs/nuscenes_models/$name.yaml --ckpt ../models/voxelnext_nuscenes_kernel1.pth
```

This is the results I tested just now as the following.

```
2023-03-23 20:30:12,459   INFO  Generate label finished(sec_per_example: 0.0482 second).                                          
2023-03-23 20:30:12,459   INFO  recall_roi_0.3: 0.000000                                                                          
2023-03-23 20:30:12,459   INFO  recall_rcnn_0.3: 0.788646                                                                         
2023-03-23 20:30:12,459   INFO  recall_roi_0.5: 0.000000                                                                          
2023-03-23 20:30:12,459   INFO  recall_rcnn_0.5: 0.563949                                                                         
2023-03-23 20:30:12,460   INFO  recall_roi_0.7: 0.000000                                                                          
2023-03-23 20:30:12,460   INFO  recall_rcnn_0.7: 0.244079                                                                         
2023-03-23 20:30:12,465   INFO  Average predicted number of objects(6019 samples): 125.095                                        

mAP: 0.6051                                                                                                                       
mATE: 0.3013                                                                                                                      
mASE: 0.2523                                                                                                                      
mAOE: 0.4057                                                                                                                      
mAVE: 0.2170                                                                                                                      
mAAE: 0.1856                                                                                                                      
NDS: 0.6664                                                                                                                       
Eval time: 106.3s                                                                                                                 

Per-class results:                                                                                                                
Object Class    AP      ATE     ASE     AOE     AVE     AAE                                                                       
car     0.839   0.186   0.156   0.166   0.232   0.205                                                                             
truck   0.555   0.327   0.182   0.174   0.198   0.221                                                                             
bus     0.705   0.338   0.182   0.097   0.377   0.236
trailer 0.381   0.596   0.201   0.629   0.164   0.181
construction_vehicle    0.210   0.703   0.433   1.218   0.123   0.300
pedestrian      0.846   0.149   0.272   0.475   0.197   0.091
motorcycle      0.628   0.199   0.235   0.321   0.310   0.236
bicycle 0.500   0.169   0.266   0.479   0.135   0.015
traffic_cone    0.694   0.143   0.325   nan     nan     nan
barrier 0.694   0.202   0.270   0.092   nan     nan
```

huihui6666, 2023-03-23:
> Hi,
> 
> Thanks for your interest in our work.
> 
> I just checked the shape of "backbone_3d.conv_input.0.weight". It is `torch.Size([16, 3, 3, 3, 5])` when I run. The input channel **5** is a default setting in nuScenes. It is defined by the nuScenes dataset config as the following line.
> 
> https://github.com/dvlab-research/VoxelNeXt/blob/50197ea102c6e69f24e6154be737cdd58d03a5f6/tools/cfgs/dataset_configs/nuscenes_dataset.yaml#L59
> 
> Would you please have a check that why your input channel is **4**, but not **5**?
> 
> Regards, Yukang Chen

Thank you for your reply！I think I know the cause of this problem. Thanks your work!

yukang2017, 2023-03-27:
Hi, I will close this issue for now. Please feel free to reopen it if you have any other issues.

closed 2023-03-27 by yukang2017

huihui6666, 2023-03-27:
> Hi, I will close this issue for now. Please feel free to reopen it if you have any other issues.

OK,thank you very much!

Wangzy-zoey, 2023-08-09:
> @yukang2017 hello，I use your work to test nuscenes datasets，but the results is not very well .And the weight is not match with the model,I use the **cbgs_voxel0075_voxelnext.yaml** to test your [...]

您好，我近期想复现voxelnext，遇到一些问题希望得到您的帮助。目前我已经下载好，但无法确定是否能正常跑模型，其次我不知道数据集该放在哪里，以及要如何测试。我是一个研一小白，期待您的回复和帮助！！万分感谢！！！

libingDY, 2023-11-16:
“Hi,

Thanks for your interest in our work.

I just checked the shape of "backbone_3d.conv_input.0.weight". It is torch.Size([16, 3, 3, 3, 5]) when I run. The input channel 5 is a default setting in nuScenes. It is defined by the nuScenes dataset config as the following line.

[VoxelNeXt/tools/cfgs/dataset_configs/nuscenes_dataset.yaml](https://github.com/dvlab-research/VoxelNeXt/blob/50197ea102c6e69f24e6154be737cdd58d03a5f6/tools/cfgs/dataset_configs/nuscenes_dataset.yaml#L59)

Line 59 in [50197ea](https://github.com/dvlab-research/VoxelNeXt/commit/50197ea102c6e69f24e6154be737cdd58d03a5f6)

 used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'], 
Would you please have a check that why your input channel is 4, but not 5?

Regards, Yukang Chen

Thank you for your reply！I think I know the cause of this problem. Thanks your work!”

libingDY, 2023-11-16:
”Hi,

Thanks for your interest in our work.

I just checked the shape of "backbone_3d.conv_input.0.weight". It is torch.Size([16, 3, 3, 3, 5]) when I run. The input channel 5 is a default setting in nuScenes. It is defined by the nuScenes dataset config as the following line.

[VoxelNeXt/tools/cfgs/dataset_configs/nuscenes_dataset.yaml](https://github.com/dvlab-research/VoxelNeXt/blob/50197ea102c6e69f24e6154be737cdd58d03a5f6/tools/cfgs/dataset_configs/nuscenes_dataset.yaml#L59)

Line 59 in [50197ea](https://github.com/dvlab-research/VoxelNeXt/commit/50197ea102c6e69f24e6154be737cdd58d03a5f6)

 used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'], 
Would you please have a check that why your input channel is 4, but not 5?

Regards, Yukang Chen

Thank you for your reply！I think I know the cause of this problem. Thanks your work!“您好请问是什么原因造成这个结果的呀？我也遇到了相同的问题。