JIA-Lab-research/VoxelNeXt #6: train nuscenes dataset use [x,y,z,inyensity] no ring

huihui6666, 2023-03-27:
Hi,yukang
I want to use the network trained by nuscenes to test my own data,because my data is including [x,y,z,intensity] but no ring ,so I want trained the network only use the four informations,but I didn't train a good network. My data config file  as follows
 I only modify the used_feature_list

POINT_FEATURE_ENCODING: {
    encoding_type: absolute_coordinates_encoding,
    #used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
    used_feature_list: ['x', 'y', 'z', 'intensity'],
    src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp'],
}

**After 20 epochs the loss of the network is very high, I dont know reason  of my problem,or maybe you can show your log of train the network.**

2023-03-25 13:41:23,834   INFO  epoch: 19/20, acc_iter=67400, cur_iter=2191/3432, batch_size=12, time_cost(epoch): 39:52/22:34, time_cost(all): 21:12:05/22:34, loss=21.39836883544922, d_time=0.01(0.04), f_time=1.10(1.06), b_time=1.11(1.09), lr=2.248708600854635e-06
2023-03-25 13:41:32,579   INFO  Save latest model to /home/mia/OpenPCDet/output/nuscenes_models/cbgs_voxel0075_voxelnext/default/ckpt/latest_model
2023-03-25 13:42:18,937   INFO  epoch: 19/20, acc_iter=67450, cur_iter=2241/3432, batch_size=12, time_cost(epoch): 40:47/21:40, time_cost(all): 21:13:00/21:40, loss=21.58650016784668, d_time=0.07(0.04), f_time=0.97(1.06), b_time=1.03(1.09), lr=2.07206874494086e-06
2023-03-25 13:43:12,857   INFO  epoch: 19/20, acc_iter=67500, cur_iter=2291/3432, batch_size=12, time_cost(epoch): 41:41/20:45, time_cost(all): 21:13:54/20:45, loss=21.092235565185547, d_time=0.01(0.04), f_time=1.05(1.06), b_time=1.06(1.09), lr=1.902672465122041e-06
2023-03-25 13:43:13,586   INFO  mroglo-202                  Sat Mar 25 13:43:13 2023  525.85.05
[0] NVIDIA GeForce RTX 4090 | 43°C,   0 % | 21645 / 24564 MB | mia(21296M)
[1] NVIDIA GeForce RTX 4090 | 48°C, 100 % | 21735 / 24564 MB | mia(21386M)
[2] NVIDIA GeForce RTX 4090 | 45°C, 100 % | 24339 / 24564 MB | mia(23990M)

2023-03-25 13:44:08,146   INFO  epoch: 19/20, acc_iter=67550, cur_iter=2341/3432, batch_size=12, time_cost(epoch): 42:36/19:50, time_cost(all): 21:14:49/19:50, loss=21.325319290161133, d_time=0.00(0.04), f_time=0.95(1.06), b_time=0.95(1.09), lr=1.7405222256555348e-06
2023-03-25 13:45:03,269   INFO  epoch: 19/20, acc_iter=67600, cur_iter=2391/3432, batch_size=12, time_cost(epoch): 43:31/18:56, time_cost(all): 21:15:44/18:56, loss=21.13665771484375, d_time=0.01(0.04), f_time=1.03(1.06), b_time=1.04(1.09), lr=1.5856203853883526e-06
2023-03-25 13:45:58,019   INFO  epoch: 19/20, acc_iter=67650, cur_iter=2441/3432, batch_size=12, time_cost(epoch): 44:26/18:02, time_cost(all): 21:16:39/18:02, loss=20.8839111328125, d_time=0.10(0.04), f_time=1.07(1.06), b_time=1.17(1.09), lr=1.4379691977227395e-06
2023-03-25 13:45:58,907   INFO  mroglo-202                  Sat Mar 25 13:45:58 2023  525.85.05
[0] NVIDIA GeForce RTX 4090 | 43°C,   0 % | 21645 / 24564 MB | mia(21296M)
[1] NVIDIA GeForce RTX 4090 | 48°C, 100 % | 21735 / 24564 MB | mia(21386M)
[2] NVIDIA GeForce RTX 4090 | 45°C, 100 % | 24339 / 24564 MB | mia(23990M)

yukang2017, 2023-03-27:
Hi Huihui,

Are you using nuScenes dataset for training?
Because in the nuScenes dataset config, the network takes 10 sweeps for each forward. If you disable the timestamps, the network will have no access to the time information of each points.

Would you please tell me what is the number of sweeps in your training?

Regards,
Yukang Chen

yukang2017, 2023-03-31:
Hi Huihui,

I think you can also try to set sweep as 1. If you don't want to use timestamp. This would also be not so good. But I think I should be better than `10 sweeps + notimestamp`.

Regards,
Yukang Chen

huihui6666, 2023-03-31:
Hi, Yukang

huihui6666, 2023-03-31:
Hi Yukang,
Thank your replying,but now I miss a problem that I train the nuscenes dataset with the **cbgs_voxel0075_voxelnext.yaml** . I use the default config and rewise nothing. 
Now the loss of hm is not converge,the result is show in the fig.But I use your pre_trained model to test and to inference is no problem,so I think the problem maybe is in the process of calculating the loss ,but I can't find the reason.
Hope for your reply for this problem, thank you.
[image]

yukang2017, 2023-04-02:
Hi Huihui,

I did not record the loss before. Would you please test the model when the model finish?

Regards,
Yukang Chen

yukang2017, 2023-04-02:
[image]

Hi Huihui,

This is the training loss I got before. I used 4 GPUs for training. This is the tensorboard file.

https://drive.google.com/file/d/16TbNtCOfygilG8SKg_Ke_B7EL-AY3tw5/view?usp=share_link

Regards,
Yukang Chen

huihui6666, 2023-04-03:
Hi yukang,
I test the model when the model finish,but the reason is not convergence and the AP is 0.
I compare my loss and your loss according the tensorborad file, I find the loss of heatmap_loss is different from yours.
The GT_heatmap you build includes 'gt-center' and 'nearst', I want to know  these two different meanings.
And I train model with the different GT_heatmap, the heatmap_loss convergence to near 0 soon when I only use the gt-center ,and the heatmap_loss always keep high when I only use the 'nearst'. 
Hope for your reply for this problem, thank you.

yukang2017, 2023-04-03:
Hi Huihui,

`draw_gaussian_to_heatmap_voxels` is to build gt heatmap based on gaussian map.

'gt-center' is to put the **gaussian center** at **the center of gt bounding boxes**.
'nearst' is to put the **gaussian center** at **the voxels whose positions are nearest** to **the the center of gt bounding boxes**.

In my previous experiments, the performance drop is slight, if I only use one of them. It is weird to see your case that it is hard to converge. I use this code and start a training yesterday. And the performance is normal. 

Would you please have a check that the config you used is the default? Would you please show me your log and config file?

Regards,
Yukang Chen

huihui6666, 2023-04-03:
Hi yukang,
Thanks for your applying soon, I use the default config file to train the model
This is the log file and config file  of training

https://drive.google.com/drive/folders/1w1zVPkcRDdo5hq9TqLsKEIor9IO9Jmvg?usp=sharing

Hope for your reply , thank you.

yukang2017, 2023-04-03:
Hi Huihui,

I did not find any errors in your config files or logs. Would you please share me your code?

Regards,
Yukang Chen

yukang2017, 2023-04-03:
Hi Huihui,

I find that in your tools/train.py. You set `args.batch_size` as 2 as a hard code, regardless of config files and numbers gpu. I think this might be the reason.

In addition, I have fixed this bug on my machine. I am retraining VoxelNeXt via your code on my machine. I will tell you my progress tomorrow.

Regards,
Yukang Chen

yukang2017, 2023-04-04:
Hi Huihui,

With the batch_size bug fixed, using your code, below is the result I got on the 1/4 nuScenes training data. I think the result is all right.

```
mAP: 0.5610                                                                                                                    
mATE: 0.3148                                                                                                                   
mASE: 0.2561                                                                                                                   
mAOE: 0.4587                                                                                                                   
mAVE: 0.2477                                                                                                                   
mAAE: 0.1847                                                                                                                   
NDS: 0.6343                                                                                                                    
Eval time: 114.7s                                                                                                              
                                                                                                                               
Per-class results:                                                                                                             
Object Class    AP      ATE     ASE     AOE     AVE     AAE                                                                    
car     0.816   0.198   0.160   0.224   0.273   0.206                                                                          
truck   0.528   0.337   0.192   0.227   0.225   0.211                                                                          
bus     0.658   0.355   0.185   0.183   0.439   0.233                                                                          
trailer 0.325   0.613   0.207   0.677   0.184   0.191                                                                          
construction_vehicle    0.186   0.727   0.433   1.128   0.133   0.309                                                          
pedestrian      0.830   0.156   0.277   0.516   0.214   0.088                                                                  
motorcycle      0.577   0.210   0.234   0.463   0.348   0.219                                                                  
bicycle 0.415   0.175   0.273   0.590   0.164   0.021                                                                          
traffic_cone    0.645   0.150   0.338   nan     nan     nan                                                                    
barrier 0.630   0.228   0.262   0.120   nan     nan                                                                            
```

You can get the trained model, log, and the tensorboard file at https://drive.google.com/file/d/1sO5kA9YEntApAMlZgMlE0UCTLW6yGOUH/view?usp=share_link

Regards,
Yukang Chen

huihui6666, 2023-04-05:
Hi yukang,

I don't understand what you mean about args.batch_size bug,in my code I set default setting of the batch_size is  2,but I can change the parameter in the script .In addition, in the log file can find the setting is 8 per gpu.

Thanks your reply.

yukang2017, 2023-04-05:
Hi Huihui,

In the `tools/train.py`, I changed the default value of `args.batch_size` from 2 to None.
In addition, please use the default batch size per GPU. I never tried 8 per GPU before.

Regards,
Yukang Chen

huihui6666, 2023-04-07:
Hi  yukang,

I still confuse about the bug of args.batch_size ,and I changed the default value as None it still didn't work. Also I build a new environment of OpenPCDet to train the network,I trained the nuscenes mini dataset. The result is as follows, obviously the heatmap loss still weird.
Thank you very much for  applying my confusion.

https://drive.google.com/drive/folders/14WeyM5Cwgt-eYNqk5aB_0vXoPUcP3_zw?usp=sharing

yukang2017, 2023-04-07:
Hi Huihui,

Would you please provide me a detailed list of your environment?
For example,  using `conda list`.

I want to know some packages about your environment and try to reproduce this bug.

Regards,
Yukang Chen

yukang2017, 2023-04-07:
Hi Huihui,

For more efficient discussion, you can directly contact me via WeChat (13261057196). 

Regards,
Yukang Chen

huihui6666, 2023-04-08:
I have already resolved this problem. The reason for the problem is due to differences of spconv library. It is recommended to use spconv versions below 2.2, such as 2.1.25.

closed 2023-04-08 by huihui6666

huixiancheng, 2024-01-10:
Hi, yukang and hui~! @yukang2017 @huihui6666 
I just reproduced this on spconv_cu113 2.2.2, switching to spconv-cu111 2.1.20 solved the problem that hm loss doesn't reduce Would you like to share the cause of the error? 
By the way, I found that with bs=1 training, the loss converges normally, and with bs>1, the loss can't be reduced. So is it correlated with the batch idx of the sparse tensor?