JIA-Lab-research/VoxelNeXt #8: Missing files?

naren2cmu, 2023-03-30:
Hi, 

The command for evaulation in README.md uses tools/scripts/dist_tesh.sh
[image]

However, dist_test.sh does not seem to be there in tools/scripts

[image]

Could you kindly help with this?

Thank you
Naren

yukang2017, 2023-03-31:
Hi Naren,

Sorry for missing them. I have uploaded them just now.

https://github.com/dvlab-research/VoxelNeXt/blob/master/tools/scripts/dist_test.sh

Regards,
Yukang Chen

yukang2017, 2023-04-02:
Hi Naren,

The test.py is [here](https://github.com/dvlab-research/VoxelNeXt/blob/master/tools/test.py). I think you should `cd tools` first.
Thanks for your remind for the broken link. I have modify the link.
https://github.com/open-mmlab/OpenPCDet/blob/master/docs/GETTING_STARTED.md

Regards,
Yukang Chen

naren2cmu, 2023-04-02:
Thanks Yukang. 

I tried that. The test.py was found this time. However, it is running into other errors. 

This is the command I ran: 

`bash scripts/dist_test.sh 1 --cfg_file /mnt/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml --ckpt /mnt/VoxelNeXt/voxelnext_nuscenes_kernel1.pth`

This is what I got. 
`+ NGPUS=1
+ PY_ARGS='--cfg_file /mnt/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml --ckpt /mnt/VoxelNeXt/voxelnext_nuscenes_kernel1.pth'
+ python -m torch.distributed.launch --nproc_per_node=1 test.py --launcher pytorch --cfg_file /mnt/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml --ckpt /mnt/VoxelNeXt/voxelnext_nuscenes_kernel1.pth
Traceback (most recent call last):
  File "test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/mnt/VoxelNeXt/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "../pcdet/__init__.py", line 4, in <module>
    from .version import __version__
ModuleNotFoundError: No module named 'pcdet.version'
Traceback (most recent call last):
  File "/opt/conda/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/opt/conda/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/opt/conda/lib/python3.7/site-packages/torch/distributed/launch.py", line 261, in <module>
    main()
  File "/opt/conda/lib/python3.7/site-packages/torch/distributed/launch.py", line 257, in main
    cmd=cmd)
subprocess.CalledProcessError: Command '['/opt/conda/bin/python', '-u', 'test.py', '--local_rank=0', '--launcher', 'pytorch', '--cfg_file', '/mnt/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml', '--ckpt', '/mnt/VoxelNeXt/voxelnext_nuscenes_kernel1.pth']' returned non-zero exit status 1.`

It seems that the pcdet.version is not being found. 

Also, you fixed the link to prepare the datasets. But I think you forgot to fix the link to install the environment. That is still broken. 
[image]

Thank you
Naren

yukang2017, 2023-04-02:
Hi Naren,

Did you install the pcdet by `python setup.py develop`?

It should be fixed if it has been installed, because of this line.
https://github.com/dvlab-research/VoxelNeXt/blob/1d67ec6daee87b730897006b248e9e4fcd05eb37/setup.py#L32

Thanks for your remind. I have updated the install document.

Regards,
Yukang Chen

closed 2023-04-06 by yukang2017