JIA-Lab-research/VoxelNeXt #9: code robustness

DezeZhao, 2023-04-02:
https://github.com/dvlab-research/VoxelNeXt/blob/c523b39ef747e5a6e8bdbcf0d05c306048e839f2/pcdet/models/dense_heads/voxelnext_head_maxpool.py#L400

I think it should add checking of `mask_cls` as follows:
                ```
if mask_cls.sum() == 0:
    continue
```

DezeZhao, 2023-04-02:
https://github.com/dvlab-research/VoxelNeXt/blob/c523b39ef747e5a6e8bdbcf0d05c306048e839f2/pcdet/models/dense_heads/voxelnext_head_maxpool.py#L69
https://github.com/dvlab-research/VoxelNeXt/blob/c523b39ef747e5a6e8bdbcf0d05c306048e839f2/pcdet/models/dense_heads/voxelnext_head_maxpool.py#L401

here `self.rectifier` is not defined, definition should be added at line 69.

yukang2017, 2023-04-02:
Thanks for your remind. Sorry for missing the `self.rectifier`. I have updated it.

closed 2023-04-03 by yukang2017