JIA-Lab-research/VoxelNeXt #10: waymo val result

DezeZhao, 2023-04-03:
[image]
@yukang2017  hi yukang! could you please tell me what config file you use for the above result on waymo val dataset marked by red box ? I reproduce the result with `voxelnext_ioubranch.yaml` for 12 epochs but it didn't be as close as the result in paper.

DezeZhao, 2023-04-03:
[image]

DezeZhao, 2023-04-03:
> [image]
> 
> @yukang2017 hi yukang! could you please tell me what config file you use for the above result on waymo val dataset marked by red box ? I reproduce the result with `voxelnext_ioubranch.yaml` for 12 epochs but it didn't be as close as the result in paper.

Or could you please share checkpoint with me?

yukang2017, 2023-04-03:
Hi Deze,

VoxelNeXt**K3** is an enhanced version of VoxelNeXt with larger model size. This config file for it is [voxelnext_ioubranch_large.yaml](https://github.com/dvlab-research/VoxelNeXt/blob/master/tools/cfgs/waymo_models/voxelnext_ioubranch_large.yaml).
Sorry for the unclearness. I will highlight this point in the README.

Please contact me via WeChat 13261057196. I will send you the checkpoint.

Regards,
Yukang Chen

yukang2017, 2023-04-03:
This is the detailed result for VoxelNeXtK3.

```
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7816 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7770 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6985 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6942 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.8147 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.7630 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.7348 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.6863
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7605 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.7490 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.7329 
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.7218
```

This is the detailed result for VoxelNeXt-2D.
```
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/AP: 0.7794 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_1/APH: 0.7747 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/AP: 0.6968 
OBJECT_TYPE_TYPE_VEHICLE_LEVEL_2/APH: 0.6925 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/AP: 0.8024 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_1/APH: 0.7347 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/AP: 0.7223 
OBJECT_TYPE_TYPE_PEDESTRIAN_LEVEL_2/APH: 0.6588
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/AP: 0.7333
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_1/APH: 0.7220
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/AP: 0.7065
OBJECT_TYPE_TYPE_CYCLIST_LEVEL_2/APH: 0.6956
```

closed 2023-04-16 by yukang2017

pycoco, 2023-09-13:
> OK thank you.

Did you reproduce the results on waymo with master configs?