JIA-Lab-research/VoxelNeXt #14: Can't run test.py successfully

Shruti-Bansal, 2023-04-09:
I have followed all the installation instructions but this is the error I get when trying to run it:

/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/VoxelNextShruti/VoxelNeXt/tools
+ NGPUS=1
+ PY_ARGS='--cfg_file cfgs/nuscenes_models/.yaml --ckpt /content/drive/MyDrive/18744_Intermediate_Demo/VoxelNeXtShruti/voxelnext_nuscenes_kernel1.pth'
+ python -m torch.distributed.launch --nproc_per_node=1 test.py --launcher pytorch --cfg_file cfgs/nuscenes_models/.yaml --ckpt /content/drive/MyDrive/18744_Intermediate_Demo/VoxelNeXtShruti/voxelnext_nuscenes_kernel1.pth
/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py:181: FutureWarning: The module torch.distributed.launch is deprecated
and will be removed in future. Use torchrun.
Note that --use-env is set by default in torchrun.
If your script expects `--local-rank` argument to be set, please
change it to read from `os.environ['LOCAL_RANK']` instead. See 
https://pytorch.org/docs/stable/distributed.html#launch-utility for 
further instructions

  warnings.warn(
usage: test.py
       [-h]
       [--cfg_file CFG_FILE]
       [--batch_size BATCH_SIZE]
       [--workers WORKERS]
       [--extra_tag EXTRA_TAG]
       [--ckpt CKPT]
       [--pretrained_model PRETRAINED_MODEL]
       [--launcher {none,pytorch,slurm}]
       [--tcp_port TCP_PORT]
       [--local_rank LOCAL_RANK]
       [--set ...]
       [--max_waiting_mins MAX_WAITING_MINS]
       [--start_epoch START_EPOCH]
       [--eval_tag EVAL_TAG]
       [--eval_all]
       [--ckpt_dir CKPT_DIR]
       [--save_to_file]
       [--infer_time]
**test.py: error: unrecognized arguments: --local-rank=0
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 2) local_rank: 0 (pid: 19545) of binary: /usr/bin/python3**
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py", line 196, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py", line 192, in main
    launch(args)
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py", line 177, in launch
    run(args)
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/run.py", line 785, in run
    elastic_launch(
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launcher/api.py", line 134, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launcher/api.py", line 250, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
test.py FAILED
------------------------------------------------------------
Failures:
  <NO_OTHER_FAILURES>
------------------------------------------------------------
Root Cause (first observed failure):
[0]:
  time      : 2023-04-09_21:34:55
  host      : 8d5d35163cb5
  rank      : 0 (local_rank: 0)
  exitcode  : 2 (pid: 19545)
  error_file: <N/A>
  traceback : To enable traceback see: https://pytorch.org/docs/stable/elastic/errors.html
============================================================

Any guidance would be helpful

yukang2017, 2023-04-10:
Hi,

If you use only 1 GPU for testing, please use this script.

`python3 test.py --cfg_file  cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml --ckpt PATH_TO_VoxelNeXt.pth
`

Regards,
Yukang Chen

Shruti-Bansal, 2023-04-13:
I tried using the script you suggested above but I am running into the below error:

warnings.warn(
Traceback (most recent call last):
  File "/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/test.py", line 14, in <module>
    from eval_utils import eval_utils
  File "/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/eval_utils/eval_utils.py", line 8, in <module>
    from pcdet.models import load_data_to_gpu
  File "/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/../pcdet/models/__init__.py", line 6, in <module>
    from .detectors import build_detector
  File "/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/../pcdet/models/detectors/__init__.py", line 1, in <module>
    from .detector3d_template import Detector3DTemplate
  File "/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/../pcdet/models/detectors/detector3d_template.py", line 6, in <module>
    from ...ops.iou3d_nms import iou3d_nms_utils
  File "/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/../pcdet/ops/iou3d_nms/iou3d_nms_utils.py", line 9, in <module>
    **from . import iou3d_nms_cuda
ImportError: cannot import name 'iou3d_nms_cuda' from 'pcdet.ops.iou3d_nms'** (/content/drive/.shortcut-targets-by-id/1JbE4k9qNvUlpfTSZhyxGbIIgEZCGZ7Ny/18744_Intermediate_Demo/LIDARDetectionVoxelNeXt/VoxelNeXt/tools/../pcdet/ops/iou3d_nms/__init__.py)
ERROR:torch.distributed.elastic.multiprocessing.api:failed (exitcode: 1) local_rank: 0 (pid: 1173) of binary: /usr/bin/python3
Traceback (most recent call last):
  File "/usr/lib/python3.9/runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.9/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py", line 196, in <module>
    main()
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py", line 192, in main
    launch(args)
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launch.py", line 177, in launch
    run(args)
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/run.py", line 785, in run
    elastic_launch(
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launcher/api.py", line 134, in __call__
    return launch_agent(self._config, self._entrypoint, list(args))
  File "/usr/local/lib/python3.9/dist-packages/torch/distributed/launcher/api.py", line 250, in launch_agent
    raise ChildFailedError(
torch.distributed.elastic.multiprocessing.errors.ChildFailedError: 
============================================================
test.py FAILED

yukang2017, 2023-04-14:
Have you installed OpenPCDet codebase?

`python3 setup.py develop`

jaaacker, 2023-05-05:
2023-05-04 20:41:27,149   INFO  **********************Start logging**********************
2023-05-04 20:41:27,149   INFO  CUDA_VISIBLE_DEVICES=ALL
2023-05-04 20:41:27,149   INFO  cfg_file         cfgs/argo2_models/cbgs_voxel01_voxelnext.yaml
2023-05-04 20:41:27,149   INFO  batch_size       16
2023-05-04 20:41:27,149   INFO  workers          4
2023-05-04 20:41:27,149   INFO  extra_tag        default
2023-05-04 20:41:27,149   INFO  ckpt             VoxelNeXt_Argo2.pth
2023-05-04 20:41:27,149   INFO  pretrained_model None
2023-05-04 20:41:27,149   INFO  launcher         none
2023-05-04 20:41:27,149   INFO  tcp_port         18888
2023-05-04 20:41:27,149   INFO  local_rank       0
2023-05-04 20:41:27,149   INFO  set_cfgs         None
2023-05-04 20:41:27,149   INFO  max_waiting_mins 30
2023-05-04 20:41:27,149   INFO  start_epoch      0
2023-05-04 20:41:27,149   INFO  eval_tag         default
2023-05-04 20:41:27,149   INFO  eval_all         False
2023-05-04 20:41:27,149   INFO  ckpt_dir         None
2023-05-04 20:41:27,149   INFO  save_to_file     False
2023-05-04 20:41:27,149   INFO  infer_time       False
2023-05-04 20:41:27,149   INFO  cfg.ROOT_DIR: /home/deevia/Argoverse/Models/VoxelNeXt-master
2023-05-04 20:41:27,149   INFO  cfg.LOCAL_RANK: 0
2023-05-04 20:41:27,149   INFO  cfg.CLASS_NAMES: ['Regular_vehicle', 'Pedestrian', 'Bicyclist', 'Motorcyclist', 'Wheeled_rider', 'Bollard', 'Construction_cone', 'Sign', 'Construction_barrel', 'Stop_sign', 'Mobile_pedestrian_crossing_sign', 'Large_vehicle', 'Bus', 'Box_truck', 'Truck', 'Vehicular_trailer', 'Truck_cab', 'School_bus', 'Articulated_bus', 'Message_board_trailer', 'Bicycle', 'Motorcycle', 'Wheeled_device', 'Wheelchair', 'Stroller', 'Dog']
2023-05-04 20:41:27,149   INFO  ----------- DATA_CONFIG -----------
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATASET: Argo2Dataset
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/argo2
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-200, -200, -20, 200, 200, 20]
2023-05-04 20:41:27,149   INFO  ----------- DATA_SPLIT -----------
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2023-05-04 20:41:27,149   INFO  ----------- INFO_PATH -----------
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['argo2_infos_train.pkl']
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['argo2_infos_val.pkl']
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.GET_ITEM_LIST: ['points']
2023-05-04 20:41:27,149   INFO  ----------- DATA_AUGMENTOR -----------
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.95, 1.05]}]
2023-05-04 20:41:27,149   INFO  ----------- POINT_FEATURE_ENCODING -----------
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity']
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity']
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': True}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.1, 0.1, 0.2], 'MAX_POINTS_PER_VOXEL': 20, 'MAX_NUMBER_OF_VOXELS': {'train': 120000, 'test': 160000}}]
2023-05-04 20:41:27,149   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/argo2_dataset.yaml
2023-05-04 20:41:27,150   INFO  ----------- MODEL -----------
2023-05-04 20:41:27,150   INFO  cfg.MODEL.NAME: VoxelNeXt
2023-05-04 20:41:27,150   INFO  ----------- VFE -----------
2023-05-04 20:41:27,150   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2023-05-04 20:41:27,150   INFO  ----------- BACKBONE_3D -----------
2023-05-04 20:41:27,150   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelResBackBone8xVoxelNeXt
2023-05-04 20:41:27,150   INFO  ----------- DENSE_HEAD -----------
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.NAME: VoxelNeXtHead
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.CLASS_AGNOSTIC: False
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.INPUT_FEATURES: 128
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.CLASS_NAMES_EACH_HEAD: [['Regular_vehicle'], ['Pedestrian', 'Bicyclist', 'Motorcyclist', 'Wheeled_rider'], ['Bollard', 'Construction_cone', 'Sign', 'Construction_barrel', 'Stop_sign', 'Mobile_pedestrian_crossing_sign'], ['Large_vehicle', 'Bus', 'Box_truck', 'Truck', 'Vehicular_trailer', 'Truck_cab', 'School_bus', 'Articulated_bus', 'Message_board_trailer'], ['Bicycle', 'Motorcycle', 'Wheeled_device', 'Wheelchair', 'Stroller'], ['Dog']]
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.KERNEL_SIZE_HEAD: 1
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.SHARED_CONV_CHANNEL: 128
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.USE_BIAS_BEFORE_NORM: True
2023-05-04 20:41:27,150   INFO  cfg.MODEL.DENSE_HEAD.NUM_HM_CONV: 2
2023-05-04 20:41:27,150   INFO  ----------- SEPARATE_HEAD_CFG -----------
[... cut ...]
  File "test.py", line 60, in eval_single_ckpt
    model.load_params_from_file(filename=args.ckpt, logger=logger, to_cpu=dist_test, 
  File "../pcdet/models/detectors/detector3d_template.py", line 368, in load_params_from_file
    model_state_disk = checkpoint['model_state']
KeyError: 'model_state'

This above error I am getting for executing the below command : 
python3 test.py --cfg_file  cfgs/argo2_models/cbgs_voxel01_voxelnext.yaml --ckpt VoxelNeXt_Argo2.pth

Where am I going wrong, please help ?

Thank you !!

yukang2017, 2023-05-07:
Hi,

Would you please try to change

`model_state_disk = checkpoint['model_state']` to `model_state_disk = checkpoint` in pcdet/models/detectors/detector3d_template.py?

Regards,
Yukang Chen

jaaacker, 2023-05-15:
Thank you so much. It worked.
But after successful run I got some files in "../output/argo2_models/" as an output.
I don't understand the significant of it. 
How can we use it for something visual output I am clue less abut it? Can you please help me out ?
Let me know if I have done something wrong...
Thank you !!