JIA-Lab-research/VoxelNeXt #16: Custom/KITTI Dataset Training Issue

hoangduyloc, 2023-04-12:
Thank you for your great work,
- I read your paper, and study your code, but I have some problem when training my custom dataset. My dataset format same as KITTI. For other backbones like PVRCNN, SECOND it works normally, but with VoxelNext It wont work and get the error bellow. It would be nice if you show me some tips to fix the error.
[image]

- It would be nice if you have the guide to train KITTI, or Custom dataset, that is really helpful.
Thank you so much.

hoangduyloc, 2023-04-14:
@yukang2017 thank you so much,

Here is the sample of VoxelNext in KITTI dataset, basically I just change the object's name, Lidar range, and the OpenPCdet will works with the data format [x, y, z, dx, dy, dz, yaw].

CLASS_NAMES: ['Car', 'Pedestrian', 'Cyclist']

DATA_CONFIG: 
    _BASE_CONFIG_: cfgs/dataset_configs/kitti_dataset.yaml
    POINT_CLOUD_RANGE: [0, -39.68, -3, 69.12, 39.68, 1]
    DATA_PROCESSOR:

    DATA_AUGMENTOR:
        DISABLE_AUG_LIST: ['placeholder']
        AUG_CONFIG_LIST:
            - NAME: gt_sampling
              USE_ROAD_PLANE: True
              DB_INFO_PATH:
                  - kitti_dbinfos_train.pkl
              PREPARE: {
                 filter_by_min_points: ['Car:5', 'Pedestrian:5', 'Cyclist:5'],
              }

              SAMPLE_GROUPS: ['Car:15','Pedestrian:15', 'Cyclist:15']
              NUM_POINT_FEATURES: 4
              DATABASE_WITH_FAKELIDAR: False
              REMOVE_EXTRA_WIDTH: [0.0, 0.0, 0.0]
              LIMIT_WHOLE_SCENE: False

            - NAME: random_world_flip
              ALONG_AXIS_LIST: ['x']

            - NAME: random_world_rotation
              WORLD_ROT_ANGLE: [-0.78539816, 0.78539816]

            - NAME: random_world_scaling
              WORLD_SCALE_RANGE: [0.95, 1.05]

MODEL:
    NAME: VoxelNeXt

    VFE:
        NAME: MeanVFE

    BACKBONE_3D:
        NAME: VoxelResBackBone8xVoxelNeXt

    DENSE_HEAD:
        NAME: VoxelNeXtHead
        CLASS_AGNOSTIC: False
        INPUT_FEATURES: 128

        CLASS_NAMES_EACH_HEAD: [
            ['car'], 
            ['truck', 'construction_vehicle'],
            ['bus', 'trailer'],
            ['barrier'],
            ['motorcycle', 'bicycle'],
            ['pedestrian', 'traffic_cone'],
        ]
        
        SHARED_CONV_CHANNEL: 128
        KERNEL_SIZE_HEAD: 1
        
        USE_BIAS_BEFORE_NORM: True
        NUM_HM_CONV: 2
        SEPARATE_HEAD_CFG:
            HEAD_ORDER: ['center', 'center_z', 'dim', 'rot', 'vel']
            HEAD_DICT: {
                'center': {'out_channels': 2, 'num_conv': 2},
                'center_z': {'out_channels': 1, 'num_conv': 2},
                'dim': {'out_channels': 3, 'num_conv': 2},
                'rot': {'out_channels': 2, 'num_conv': 2},
                'vel': {'out_channels': 2, 'num_conv': 2},
            }

        TARGET_ASSIGNER_CONFIG:
            FEATURE_MAP_STRIDE: 8
            NUM_MAX_OBJS: 500
            GAUSSIAN_OVERLAP: 0.1
            MIN_RADIUS: 2

        LOSS_CONFIG:
            LOSS_WEIGHTS: {
                'cls_weight': 1.0,
                'loc_weight': 0.25,
                'code_weights': [1.0, 1.0, 1.0, 1.0, 1.0, 1.0, 0.2, 0.2, 1.0, 1.0]
            }

        POST_PROCESSING:
            SCORE_THRESH: 0.1
            POST_CENTER_LIMIT_RANGE: [-61.2, -61.2, -10.0, 61.2, 61.2, 10.0]
            MAX_OBJ_PER_SAMPLE: 500
            NMS_CONFIG:
                NMS_TYPE: nms_gpu
                NMS_THRESH: 0.2
                NMS_PRE_MAXSIZE: 1000
                NMS_POST_MAXSIZE: 83

    POST_PROCESSING:
        RECALL_THRESH_LIST: [0.3, 0.5, 0.7]

        EVAL_METRIC: kitti

OPTIMIZATION:
    BATCH_SIZE_PER_GPU: 4
    NUM_EPOCHS: 20

    OPTIMIZER: adam_onecycle
    LR: 0.001
    WEIGHT_DECAY: 0.01
    MOMENTUM: 0.9

    MOMS: [0.95, 0.85]
    PCT_START: 0.4
    DIV_FACTOR: 10
    DECAY_STEP_LIST: [35, 45]
    LR_DECAY: 0.1
    LR_CLIP: 0.0000001

    LR_WARMUP: False
    WARMUP_EPOCH: 1

    GRAD_NORM_CLIP: 10
    
    Actually, in my previous experience, if it works in KITTI dataset, then it should work with Custom dataset well.
    I'm looking forward to your reply, thank you

yukang2017, 2023-04-16:
Hi @hoangduyloc,

Please try this config.

https://github.com/dvlab-research/VoxelNeXt/blob/master/tools/cfgs/kitti_models/voxelnext.yaml

I have trained and tested it on KITTI. Below is the result. Note that, you can try to tune the loss weight and post-processing hyper-parameters to obtained better result.

```
bbox AP:90.3782, 89.3870, 88.7935
bev  AP:89.1411, 86.9760, 85.4110
3d   AP:87.0420, 77.2991, 76.5183
aos  AP:90.37, 89.25, 88.61
Car AP_R40@0.70, 0.70, 0.70:
bbox AP:95.3763, 93.7284, 91.4338
bev  AP:91.5849, 87.5852, 86.8621
3d   AP:88.6480, 80.1391, 77.7470
aos  AP:95.36, 93.56, 91.23
Car AP@0.70, 0.50, 0.50:
bbox AP:90.3782, 89.3870, 88.7935
bev  AP:94.1812, 93.1726, 89.0778
3d   AP:94.1367, 89.4471, 88.9638
aos  AP:90.37, 89.25, 88.61
Car AP_R40@0.70, 0.50, 0.50:
bbox AP:95.3763, 93.7284, 91.4338
bev  AP:96.4958, 95.4601, 93.7758
3d   AP:96.3723, 94.0739, 93.5335
aos  AP:95.36, 93.56, 91.23

Pedestrian AP@0.50, 0.50, 0.50:
bbox AP:78.9076, 74.8837, 71.7826
bev  AP:63.9831, 60.5570, 56.8279
3d   AP:59.9474, 55.8261, 51.2499
aos  AP:77.47, 73.07, 69.42
Pedestrian AP_R40@0.50, 0.50, 0.50:
bbox AP:79.7630, 76.3886, 73.1756
bev  AP:64.0781, 60.2438, 55.8577
3d   AP:58.5884, 54.7600, 49.9538
aos  AP:78.20, 74.33, 70.46
Pedestrian AP@0.50, 0.25, 0.25:
bbox AP:78.9076, 74.8837, 71.7826
bev  AP:84.4825, 81.7914, 79.0559
3d   AP:84.2736, 81.5116, 78.7618
aos  AP:77.47, 73.07, 69.42
Pedestrian AP_R40@0.50, 0.25, 0.25:
bbox AP:79.7630, 76.3886, 73.1756
bev  AP:86.4864, 84.1681, 80.5073
3d   AP:86.2521, 83.8222, 80.1752
aos  AP:78.20, 74.33, 70.46

Cyclist AP@0.50, 0.50, 0.50:
bbox AP:88.5463, 75.0048, 72.4298
bev  AP:87.0297, 70.7931, 66.7646
3d   AP:83.7304, 67.3986, 62.8305
aos  AP:88.39, 74.64, 72.07
Cyclist AP_R40@0.50, 0.50, 0.50:
bbox AP:93.1879, 76.7464, 72.9173
bev  AP:91.1662, 71.6552, 67.2355
3d   AP:87.1778, 67.1069, 63.0479
aos  AP:93.00, 76.35, 72.52
Cyclist AP@0.50, 0.25, 0.25:
bbox AP:88.5463, 75.0048, 72.4298
bev  AP:88.3133, 74.7221, 70.6009
3d   AP:88.3133, 74.7146, 70.5426
aos  AP:88.39, 74.64, 72.07
Cyclist AP_R40@0.50, 0.25, 0.25:
bbox AP:93.1879, 76.7464, 72.9173
bev  AP:92.7491, 75.4853, 71.3751
3d   AP:92.7467, 75.4782, 71.2855
aos  AP:93.00, 76.35, 72.52

```

Regards,
Yukon Chen

hoangduyloc, 2023-04-17:
Thank you so much! I will test it out

yukang2017, 2023-04-18:
Thanks. I will close this issue. Please feel free to reopen it, if there are any issue.

closed 2023-04-18 by yukang2017