JIA-Lab-research/VoxelNeXt #20: Question about train time on waymo

sky-fly97, 2023-04-18:
Hello, can you share your machines and training cost time on waymo? Such as `waymo_models/voxelnext2d_ioubranch.yaml` and `waymo_models/voxelnext_ioubranch_large.yaml)`

sky-fly97, 2023-04-18:
I have did experiments on 4 A100 and 20% waymo data, it seems that the k3 version `voxelnext_ioubranch_large.yaml` consumes twice as much time as the 2D version `voxelnext2d_ioubranch.yaml`. The former is 10hour, the latter is 4.5hour

yukang2017, 2023-04-18:
Hi @sky-fly97 ,

My machine (4 GPUs V100) is very very old... Its CPU for loading data really drives me crazy. It took me about 5 days to train `voxelnext2d_ioubranch.yaml` and 1 week to train `voxelnext_ioubranch_large.yaml`.

Regards,
Yukang Chen

closed 2023-04-18 by sky-fly97

seonhoon1002, 2023-06-19:
> Hi @sky-fly97 ,
> 
> My machine (4 GPUs V100) is very very old... Its CPU for loading data really drives me crazy. It took me about 5 days to train `voxelnext2d_ioubranch.yaml` and 1 week to train `voxelnext_ioubranch_large.yaml`.
> 
> Regards, Yukang Chen

Can I ask how many epochs you set when you train on 20% waymo dataset. 
Because In my case Its converge time is so long and performance is nan when I use 20% waymo dataset

yukang2017, 2023-06-19:
Hi,

In my 20% training, I used the exactly same settings to the full dataset, both 12 epochs (8 GPUs).
It is weird. The training results for 100% and 20% data should not have such gap.

Regards,
Yukang Chen

seonhoon1002, 2023-06-23:
Hello,

I solved the problem.

I actually reimplemented your code based on mmdetection3D. 

https://github.com/dvlab-research/VoxelNeXt/blob/b5b7d393cd1d0ecbbaeaca365b453b488791035d/pcdet/utils/loss_utils.py#L421

This class might have some problems because this Loss class has no function of splitting batches compared with other Loss classes such as RegLossSparse or something else. 

I think I didn't find a similar function for that in your code.

Finally, I got below performances in Waymo Datasets based on your "voxelnext_ioubranch_large.yaml"

Vehicle/L2 mAPH: 0.6657
Pedestrian/L2 mAPH: 0.6599, 
Cyclist/L2 mAPH: 0.7042

csinsgcc, 2023-10-03:
> Hello,
>
> I solved the problem.
>
> I actually reimplemented your code based on mmdetection3D. [...]

Hi,

I have also been using mmdetection3d to reproduce recently, and I have also encountered a problem with long convergence time. 

I have analyzed the loss and guess it may be a problem with focal loss. 

May I ask how you resolved the problem with focal loss?

seonhoon1002, 2023-10-03:
> Hi,
>
> I have also been using mmdetection3d to reproduce recently, and I have also encountered a problem with long convergence time.
>
> I have analyzed the loss and guess it may be a problem with focal [...]

Hello, 

In my perspective, I think your problem is not caused by focal loss

In VoxelNeXt, to make targets, they use "for loop" for all ground truth 

and It gets worse when you use multi-task head groups such as ArgoverseV2, because we should make targets for each task per ground truth.

I tried to make a faster targeting module, but this repository uses Gaussian Focal Loss and nearest assignments, so I can't resolve the convergence problem. Because I don't know how avoid "for loop" when I implement GFL and nearest assignment 

Maybe someday, If someone makes "cuda version" targeting module, we can make faster training convergence.

Good luck

csinsgcc, 2023-10-04:
> Hello,
>
> In my perspective, I think your problem is not caused by focal loss
>
> In VoxelNeXt, to make targets, they use "for loop" for all ground truth
>
> and It gets worse when you use multi-task head [...]

Thank you very much for your detailed reply~

seonhoon1002, 2024-01-15:
> currence.
> 
> Thanks.

Hello,

Unfortunately, I can't remember the exact evaluation cost time. It's a too long time... 

But it is weird that the final results are several times longer than the evaluation time. 

I'm sorry I can't help.

Good luck!

Rzlnmc, 2024-01-15:
> > currence.
> > Thanks.
> 
> Hello,
> 
> Unfortunately, I can't remember the exact evaluation cost time. It's a too long time...
> 
> But it is weird that the final results are several times longer than the evaluation time.
> 
> I'm sorry I can't help.
> 
> Good luck!

Thank you for your prompt response! 

I am using the official configuration files and code from openpcdet for training, but the time spent on generating results is four times longer than the validation time. 

I have found some [answers](https://github.com/open-mmlab/OpenPCDet/issues/1146#issuecomment-1275639697), but they haven't resolved the issue. 

Anyway, thank you very much."

Wansit99, 2024-09-10:
> Hello,
>
> In my perspective, I think your problem is not caused by focal loss
>
> In VoxelNeXt, to make targets, they use "for loop" for all ground truth
>
> and It gets worse when you use multi-task head [...]

I am currently working on reproducing this paper using MM3D. Could you please share your reproduction code if it's available? I would greatly appreciate it, and I'm also open to purchasing it if needed. Thank you very much!

seonhoon1002, 2024-09-11:
> I am currently working on reproducing this paper using MM3D. Could you please share your reproduction code if it's available? I would greatly appreciate it, and I'm also open to purchasing it if [...]

I'm afraid I can't share the code, because it's a long time since I reimplemented this code. And I don't think It's a good idea to reproduce this repository by MM3D because there are so many important hyper-parameter that we don't know and only available at openpcdet. Nevertheless, I wish your trials are successful, and it would be great if you could share the code.