JIA-Lab-research/VoxelNeXt #21: About the FLOPs.

AlmoonYsl, 2023-04-18:
Hi, thanks for your excellent work.
As shown in Tab. 1, the FLOPs is calculated for sparse cnn and head separately. I noticed that the FLOPs of sparse cnn is 62.9G, which should include the 3D Backbone. So, I want to know whether the sparse conv in 3D backbone is considered in calculation. AFAK, the mmcv and fvcore do not support the FLOPs calculation for sparse conv.

AlmoonYsl, 2023-04-18:
> Hi,
> 
> spconv is included in FLOPs calculation. We use the FLOPs calculation tool in https://github.com/CVMI-Lab/SparseKD

Got it. Thanks for your reply!

closed 2023-04-18 by AlmoonYsl

yBeOne, 2023-07-04:
Where is the FLOPs calculation tool in SparseKD?

yukang2017, 2023-07-05:
Please refer to this.

https://github.com/CVMI-Lab/SparseKD/blob/8572093edc77d4217f7596d9e1de68c6dc77e420/pcdet/utils/common_utils.py#L344