JIA-Lab-research/VoxelNeXt #25: waymo results reference

DezeZhao, 2023-05-25:
[image]

hi @yukang2017 , could you please show me where the results of CenterPoint and SST you refer?
for CenterPoint, original paper results are not the same as you provided, for SST, I can only find Veh. and Ped. results from the original paper. Looking forward to your reply.

yukang2017, 2023-05-25:
Hi,

I refer these from this paper.

Fully Sparse 3D Object Detection, NeurIPS 2022.
https://arxiv.org/pdf/2207.10035.pdf

Regards,
Yukang Chen

DezeZhao, 2023-05-25:
OK  thank you! I get it.

DezeZhao, 2023-05-25:
Another question: Have you ever tried to train voxelnext_ioubranch and voxelnext_ioubranch_large configs with 30 epochs to get even better results?

DezeZhao, 2023-05-25:
yeah, thank you for your reply in time.

closed 2023-05-25 by DezeZhao