JIA-Lab-research/VoxelNeXt #26: How to visualize the effective receptive fields (ERFs)?

Raiden-cn, 2023-05-25:
Hi, could give me some advice about visualizing the effective receptive field (like Fig. 5 in the paper) and normal receptive field?
Thanks!

yukang2017, 2023-05-25:
Hi,

I follow this paper for the visualization of effective receptive fields. I will ask @X-Lai  for the code sharing about it.

Stratified Transformer for 3D Point Cloud Segmentation, CVPR 2023.
https://openaccess.thecvf.com/content/CVPR2022/papers/Lai_Stratified_Transformer_for_3D_Point_Cloud_Segmentation_CVPR_2022_paper.pdf

Regards,
Yukang Chen

Raiden-cn, 2023-05-26:
Thanks for your reply.
I'm looking forward to the code!

X-Lai, 2023-05-28:
@Raiden-cn Hi, for ERF visualization, please follow the below code.
```
activation = output[select_idx].sum()
activation.backward()
grad = feat.grad.norm(dim=-1).clone()
grad = (grad - grad.min()) / (grad.max() - grad.min())
grad = grad.detach().cpu().numpy()
grad = np.clip(grad * 10, 0, 1)
grad = (grad * 255).astype(np.uint8)
cmap = cv2.cvtColor(cv2.applyColorMap(grad[:, None], cv2.COLORMAP_JET), cv2.COLOR_BGR2RGB)
```
where the `output` is the output features with the shape [N, C], and `select_idx` is the index of the selected feature.

Raiden-cn, 2023-05-28:
@X-Lai Get it! Thx!
Should I use a trained checkpoint to execute or at any epoch?

X-Lai, 2023-05-28:
Yes, it is based on the model inference process. And don't forget to first make the `requires_grad` of input tensor to `True`:)

closed 2023-06-14 by yukang2017

yinheyanxian, 2024-03-11:
> @X-Lai Get it! Thx! Should I use a trained checkpoint to execute or at any epoch?
Have you finished the erf visualization for voxelnext? Can you tell me where the code given by X-Lai should be put?

yinheyanxian, 2024-03-15:
> @Raiden-cn Hi, for ERF visualization, please follow the below code.
> 
> ```
> activation = output[select_idx].sum()
> activation.backward()
> grad = feat.grad.norm(dim=-1).clone()
> grad = (grad - grad.min()) / (grad.max() - grad.min())
> grad = grad.detach().cpu().numpy()
> grad = np.clip(grad * 10, 0, 1)
> grad = (grad * 255).astype(np.uint8)
> cmap = cv2.cvtColor(cv2.applyColorMap(grad[:, None], cv2.COLORMAP_JET), cv2.COLOR_BGR2RGB)
> ```
> 
> where the is the output features with the shape [N, C], and is the index of the selected feature.`output``select_idx`

Thank you for this wonderful piece of code. But I have one more question. What does feat on the third line ( grad = feat.grad.norm(dim=-1).clone() ) mean?