JIA-Lab-research/VoxelNeXt #30: Question about your code

seonhoon1002, 2023-06-13:
Hello, 
I have a great time exploring your code.

But  I have a question which I can't understand

https://github.com/dvlab-research/VoxelNeXt/blob/b5b7d393cd1d0ecbbaeaca365b453b488791035d/pcdet/models/dense_heads/voxelnext_head.py#L508-L521

518 line in the above code, you change spatial_indices change [2,1]
But I don't understand why you change the order. 
Does the order affect some compatibilities of other libraries such as nuscenes or waymo?
Or is it ok changing [2,1] --> [1,2] if I modify "def _get_predicted_boxes" and " def assign_target_of_single_head" to encode and decode bboxes

Thank you for reading.

yukang2017, 2023-06-14:
Hi,

This is because the different order of (x, y) coordinates between the codebase of spconv and openpcdet.

In spconv, the indices dim order by default is (y, x).
In openpcdet, the order to compute loss is (x, y). I change this to keep them the same order to be (x, y).
I think it is OK to change the order in other functions, but this way might be convenient enough.

Regards,
Yukang Chen

seonhoon1002, 2023-06-14:
I understand.

Thank you for your reply.

closed 2023-06-14 by seonhoon1002