JIA-Lab-research/VoxelNeXt #31: Question about cfgs file

Seong-min-Park, 2023-06-16:
Thank you for the excellent research. 
I have a question about the cfgs file provided in the code. When I tried to reproduce using the [cbgs_voxel0075_voxelnext.yaml](https://github.com/dvlab-research/VoxelNeXt/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml), I noticed that the FLOPs calculation shows a model with a spatial pruning ratio of 0, as seen in Table 3. Is there a model config file available with a spatial pruning ratio of 0.5, or is there a way to separately configure the pruning ratio?

yukang2017, 2023-06-16:
You can change the backbone name to be "VoxelResBackBone8xVoxelNeXtSPS" in the config file.

Seong-min-Park, 2023-06-19:
Thank you for your response. I have an additional question. To obtain a model with 33.6 GFLOPs, similar to the result with a ratio of 0.5 in table3, should I set the pruning ratio of the SPS module to 0.5? (When I tried reproducing it, setting the pruning ratio of SPS module to 0.5 didn't result in the expected reduction in computational cost.)

yukang2017, 2023-06-19:
Yes. 0.5 is the right setting. Would you please show me how you calculate FLOPs?

I used the method provided in this work.
https://github.com/CVMI-Lab/SparseKD

Seong-min-Park, 2023-06-20:
I measured the computation cost of the model by directly calculating the number of active points in each layer using _**input.features.shape[0]**_ in the conv forward of the spconv library. The flops of each layer were computed as _**input_channel * output_channel * kernel^dim * #active_points**_, and then I summed up the flops of all layers.