JIA-Lab-research/VoxelNeXt #33: Question on NMS configs in the POST_PROCESSING

Galaxy-ZRX, 2023-06-27:
Hi,

Thank you for the impressive work! I noticed that you mentioned in your paper that VoxelNext doesn't require NMS process, but I still found the NMS settings in the POST_PROCESSING part of the configs. May I ask what is the purpose of this setting or if I misunderstand it?

Many thanks,
Ruixiao

Galaxy-ZRX, 2023-06-27:
Hi,

After checking the code details, it seems that you rewrote the post-processing function in the detector file, and the nms process is removed, which means that the NMS settings in the configs are actually never used. Hope that you can confirm if I understand this correctly or not and close the question, thank you! :)

yukang2017, 2023-06-28:
Hi,

Thanks for your interest in our work. 

NMS can be replace by sparse max pooling. You can refer to this config, where NMS is never used.

https://github.com/dvlab-research/VoxelNeXt/blob/master/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext_maxpool.yaml

Regards,
Yukang Chen

csinsgcc, 2023-06-28:
Hi，

I compared the two configs you provided, but the difference between them is DENSE_HEAD name and SIZE_EACH_HEAD. The NMS_CONFIG part was used in both configs and they are exactly the same. So, as Ruixiao said, the function was rewritten, so NMS_CONFIG part in maxpool.yaml are actually never used? 

But I don't understand which part of the function was rewritten?

Thank you!

yukang2017, 2023-06-28:
Hi,

Yes. NMS_CONFIG in the maxpool.yaml is actually never used.

The max pooling function is written [here](https://github.com/dvlab-research/spconv-plus/blob/c39c7ff84c8aef4a5074b510cc6949caf9b20a8b/spconv/pytorch/pool.py#L399).

Regards,
Yukang Chen

Galaxy-ZRX, 2023-06-28:
Hi Yukang,

Thank you for your reply and explanation! I quickly checked the related head files and found that they are quite different. As you said, the maxpool-version head didn't use the NMS parameters while the VoxelNeXtHead uses them and their forward functions are quite different. I should focus more on the maxpool version which is the main design of your VoxelNext as I understand. Thank you for your reply again!

yukang2017, 2023-06-29:
Great.

closed 2023-06-29 by yukang2017

csinsgcc, 2023-06-29:
Thank you very much for your reply! I also understand.