JIA-Lab-research/VoxelNeXt #34: Question about training time on NuScenes

dabi4256, 2023-06-30:
您好，我自己用完整的nuScenes数据集训练，显卡是两张3090， BATCH_SIZE_PER_GPU=8，用的是‘cbgs_voxel0075_voxelnext.yaml’配置文件。训练时间稳定的话大概是250个小时，这是正常的吗。

yukang2017, 2023-06-30:
你好，感觉不是很正常。我用4卡V100，BATCH_SIZE_PER_GPU=4. 训练大概2天多点。

dabi4256, 2023-07-01:
你好，感谢你的回复。
请问这个时间是实际跑完需要2天，还是程序给出的预估时间。 我代码还没有训练完，只是看到程序给出的时间是200个小时。
而且，我用双卡3090跟单卡3090的预估时间差别不大。

yukang2017, 2023-07-01:
你好，我预估时间和实际时间基本差不多。

learnuser1, 2023-07-06:
> ```
>     您好，我自己用完整的nuScenes数据集训练，显卡是两张3090， BATCH_SIZE_PER_GPU=8，用的是‘cbgs_voxel0075_voxelnext.yaml’配置文件。训练时间稳定的话大概是200个小时，这是正常的吗。
> ```

> 

你好，这个自己训练的话效果跟原论文比的话怎么样。然后这个模型跟centerpoint比的话要好很多嘛

dabi4256, 2023-07-06:
> > ```
> >     您好，我自己用完整的nuScenes数据集训练，显卡是两张3090， BATCH_SIZE_PER_GPU=8，用的是‘cbgs_voxel0075_voxelnext.yaml’配置文件。训练时间稳定的话大概是200个小时，这是正常的吗。
> > ```
> 
> > 
> 
> 你好，这个自己训练的话效果跟原论文比的话怎么样。然后这个模型跟centerpoint比的话要好很多嘛

我的机器跑的很慢，20个epoch还没跑完。但是我测试了第10个epoch，mAP已经到了59.32了。CenterPoint我没跑过，但是论文里面写的是CenterPoint的mAP为 58.6.

learnuser1, 2023-07-07:
> > > ```
> > >     您好，我自己用完整的nuScenes数据集训练，显卡是两张3090， BATCH_SIZE_PER_GPU=8，用的是‘cbgs_voxel0075_voxelnext.yaml’配置文件。训练时间稳定的话大概是200个小时，这是正常的吗。
> > > ```
> > 
> > 
> > > 
> > 
> > 
> > 你好，这个自己训练的话效果跟原论文比的话怎么样。然后这个模型跟centerpoint比的话要好很多嘛
> 
> 我的机器跑的很慢，20个epoch还没跑完。但是我测试了第10个epoch，mAP已经到了59.32了。CenterPoint我没跑过，但是论文里面写的是CenterPoint的mAP为 58.6.

好的，感谢

1806610292, 2024-08-09:
> ```
>     您好，我自己用完整的nuScenes数据集训练，显卡是两张3090， BATCH_SIZE_PER_GPU=8，用的是‘cbgs_voxel0075_voxelnext.yaml’配置文件。训练时间稳定的话大概是200个小时，这是正常的吗。
> ```

你好，请问这个问题解决了吗，我也遇到了同样的问题，训练时间非常长！