JIA-Lab-research/VoxelNeXt #35: ONNX export

yBeOne, 2023-07-04:
RuntimeError: Only tuples, lists and Variables are supported as JIT inputs/outputs. Dictionaries and strings are also accepted, but their usage is not recommended. Here, received an input of unsupported type: numpy.ndarray.
Have you export VoxelNeXt to ONNX?

yukang2017, 2023-07-05:
Hi,

I think this might be hard. Because the spconv backbone receives sparse tensors, which is not supported yet by ONNX.

I have converted the head of VoxelNeXt to ONNX. That is easy. Please refer to this one.

Regards,
Yukang Chen

HuangVictorAuto, 2023-07-13:
> Hi,
> 
> I think this might be hard. Because the spconv backbone receives sparse tensors, which is not supported yet by ONNX.
> 
> I have converted the head of VoxelNeXt to ONNX. That is easy. Please refer to this one.
> 
> Regards, Yukang Chen

Hi, Dear Yukang

can you share where is the link or postion for "this one"?

Thanks!

ArseniuML, 2023-09-27:
> I think this might be hard. Because the spconv backbone receives sparse tensors, which is not supported yet by ONNX.

Could it be done via this plugin?

https://github.com/jingyue202205/SE-SSD-AI-TRT/blob/master/sparseConv3dlayer.h

Gaussiandra, 2024-04-19:
> Hi,
> 
> I think this might be hard. Because the spconv backbone receives sparse tensors, which is not supported yet by ONNX.
> 
> I have converted the head of VoxelNeXt to ONNX. That is easy. Please refer to this one.
> 
> Regards, Yukang Chen

Could you please provide config to test ONNX model?