JIA-Lab-research/VoxelNeXt #42: [Question] How to run a single inference on this model?

MyronRodrigues-StreetDrone, 2023-08-08:
Thank you for open sourcing your work.

I need some help to run a single inference on the model. Evaluation results match the results shown in readme.

I managed to get 10 Lidar seeps from the nuscenes dataset to pass into the data processor how do I run the model on batch size 1?

I get results with a very low score.

```
[{'pred_boxes': tensor([[ 5.9413e+00,  3.6396e+00,  3.4623e-02,  ...,  2.1592e+00,
          8.4747e-02, -4.7400e-02],
        [ 8.3678e+00,  2.4014e+00, -1.2982e-02,  ...,  1.9676e+00,
          1.6921e-02,  3.2265e-02],
        [ 7.1588e+00,  3.0267e+00,  8.0288e-03,  ..., -7.9587e-01,
         -1.0607e-02,  4.5947e-02],
        ...,
        [-3.2396e+01,  2.6405e+01,  1.2968e-02,  ..., -1.3317e+00,
          2.1994e-02,  1.7819e-02],
        [ 9.5994e+00,  5.0404e+01,  5.9437e-03,  ..., -1.0493e+00,
          1.6636e-02,  8.8143e-03],
        [-7.8000e+00, -8.9908e+00,  4.4057e-03,  ..., -1.0186e+00,
          4.8190e-02,  5.5851e-03]], device='cuda:0'),
  'pred_ious': [None, None, None, None, None, None],
  'pred_labels': tensor([1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
        1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3,
        3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3,
        3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3,
        3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 3, 5, 5,
        5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5,
        5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5,
        5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5, 5,
        5, 5, 5, 5, 5, 5, 5, 5, 5, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6,
        6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6,
        6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6,
        6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 6, 7, 9, 9, 9,
        9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9,
        9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9,
        9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9, 9,
        9, 9, 9, 9, 9, 9, 9, 9], device='cuda:0'),
  'pred_scores': tensor([0.1018, 0.1016, 0.1016, 0.1015, 0.1015, 0.1014, 0.1013, 0.1013, 0.1013,
        0.1013, 0.1013, 0.1013, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012,
        0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012,
        0.1012, 0.1012, 0.1012, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011,
        0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011,
        0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011,
        0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011,
        0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011,
        0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011, 0.1011,
        0.1011, 0.1011, 0.1045, 0.1034, 0.1034, 0.1033, 0.1032, 0.1032, 0.1032,
        0.1031, 0.1031, 0.1031, 0.1031, 0.1031, 0.1030, 0.1030, 0.1030, 0.1030,
        0.1030, 0.1030, 0.1030, 0.1030, 0.1029, 0.1029, 0.1029, 0.1029, 0.1029,
        0.1029, 0.1029, 0.1029, 0.1029, 0.1029, 0.1029, 0.1029, 0.1029, 0.1029,
        0.1029, 0.1029, 0.1029, 0.1029, 0.1029, 0.1028, 0.1028, 0.1028, 0.1028,
        0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028,
        0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028,
        0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028,
        0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028, 0.1028,
        0.1028, 0.1028, 0.1028, 0.1028, 0.1032, 0.1026, 0.1023, 0.1022, 0.1021,
        0.1019, 0.1019, 0.1019, 0.1018, 0.1018, 0.1018, 0.1017, 0.1017, 0.1017,
        0.1017, 0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1015, 0.1015, 0.1015,
        0.1015, 0.1015, 0.1015, 0.1015, 0.1014, 0.1014, 0.1014, 0.1014, 0.1014,
        0.1014, 0.1014, 0.1014, 0.1014, 0.1014, 0.1014, 0.1014, 0.1014, 0.1014,
        0.1014, 0.1013, 0.1013, 0.1013, 0.1013, 0.1013, 0.1013, 0.1013, 0.1013,
        0.1013, 0.1013, 0.1013, 0.1013, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012,
        0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012,
        0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012,
        0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1012, 0.1026, 0.1022, 0.1022,
        0.1021, 0.1021, 0.1020, 0.1020, 0.1020, 0.1019, 0.1019, 0.1018, 0.1018,
        0.1018, 0.1017, 0.1017, 0.1017, 0.1017, 0.1016, 0.1016, 0.1016, 0.1016,
        0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1016,
        0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1016, 0.1015, 0.1015, 0.1015,
        0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015,
        0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015,
        0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015,
        0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015,
        0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1015, 0.1000,
        0.1039, 0.1037, 0.1036, 0.1034, 0.1034, 0.1034, 0.1034, 0.1034, 0.1034,
        0.1034, 0.1034, 0.1034, 0.1034, 0.1034, 0.1034, 0.1033, 0.1033, 0.1033,
        0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033,
        0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033,
        0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033,
        0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033,
        0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1033, 0.1032, 0.1032,
        0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032,
        0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032, 0.1032,
        0.1032, 0.1032], device='cuda:0')}]
```

AmrinKareem, 2023-08-08:
+1 
I need help with this too.

Wangzy-zoey, 2023-08-10:
hello，how do you train the datasets? I’m a freshman to run the work, please give me some tips.Thanks!