JIA-Lab-research/VoxelNeXt #44: How to use the NuScenes dataset? Please!!!

Wangzy-zoey, 2023-08-09:
~/3Detector/VoxelNeXt/tools:   ### python3 test.py --cfg_file /home/wzy/3Detector/VoxelNeXt/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml --ckpt /home/wzy/3Detector/voxelnext_nuscenes_kernel1.pth

2023-08-09 20:08:21,915   INFO  **********************Start logging**********************
2023-08-09 20:08:21,915   INFO  CUDA_VISIBLE_DEVICES=ALL
2023-08-09 20:08:21,915   INFO  cfg_file         /home/wzy/3Detector/VoxelNeXt/tools/cfgs/nuscenes_models/cbgs_voxel0075_voxelnext.yaml
2023-08-09 20:08:21,915   INFO  batch_size       4
2023-08-09 20:08:21,915   INFO  workers          4
2023-08-09 20:08:21,915   INFO  extra_tag        default
2023-08-09 20:08:21,915   INFO  ckpt             /home/wzy/3Detector/voxelnext_nuscenes_kernel1.pth
2023-08-09 20:08:21,915   INFO  pretrained_model None
2023-08-09 20:08:21,915   INFO  launcher         none
2023-08-09 20:08:21,915   INFO  tcp_port         18888
2023-08-09 20:08:21,915   INFO  local_rank       0
2023-08-09 20:08:21,915   INFO  set_cfgs         None
2023-08-09 20:08:21,915   INFO  max_waiting_mins 30
2023-08-09 20:08:21,915   INFO  start_epoch      0
2023-08-09 20:08:21,915   INFO  eval_tag         default
2023-08-09 20:08:21,915   INFO  eval_all         False
2023-08-09 20:08:21,915   INFO  ckpt_dir         None
2023-08-09 20:08:21,915   INFO  save_to_file     False
2023-08-09 20:08:21,915   INFO  infer_time       False
2023-08-09 20:08:21,915   INFO  cfg.ROOT_DIR: /home/wzy/3Detector/VoxelNeXt
2023-08-09 20:08:21,915   INFO  cfg.LOCAL_RANK: 0
2023-08-09 20:08:21,915   INFO  cfg.CLASS_NAMES: ['car', 'truck', 'construction_vehicle', 'bus', 'trailer', 'barrier', 'motorcycle', 'bicycle', 'pedestrian', 'traffic_cone']
2023-08-09 20:08:21,915   INFO  ----------- DATA_CONFIG -----------
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATASET: NuScenesDataset
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATA_PATH: ../data/nuscenes
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.VERSION: v1.0-mini
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.MAX_SWEEPS: 10
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.PRED_VELOCITY: True
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.SET_NAN_VELOCITY_TO_ZEROS: True
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.FILTER_MIN_POINTS_IN_GT: 1
2023-08-09 20:08:21,915   INFO  ----------- DATA_SPLIT -----------
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATA_SPLIT.train: train
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATA_SPLIT.test: val
2023-08-09 20:08:21,915   INFO  ----------- INFO_PATH -----------
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.INFO_PATH.train: ['nuscenes_infos_10sweeps_train.pkl']
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.INFO_PATH.test: ['nuscenes_infos_10sweeps_val.pkl']
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.POINT_CLOUD_RANGE: [-54.0, -54.0, -5.0, 54.0, 54.0, 3.0]
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.BALANCED_RESAMPLING: True
2023-08-09 20:08:21,915   INFO  ----------- DATA_AUGMENTOR -----------
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.DISABLE_AUG_LIST: ['placeholder']
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATA_AUGMENTOR.AUG_CONFIG_LIST: [{'NAME': 'gt_sampling', 'DB_INFO_PATH': ['nuscenes_dbinfos_10sweeps_withvelo.pkl'], 'USE_SHARED_MEMORY': False, 'DB_DATA_PATH': ['nuscenes_dbinfos_10sweeps_withvelo_global.pkl.npy'], 'PREPARE': {'filter_by_min_points': ['car:5', 'truck:5', 'construction_vehicle:5', 'bus:5', 'trailer:5', 'barrier:5', 'motorcycle:5', 'bicycle:5', 'pedestrian:5', 'traffic_cone:5']}, 'SAMPLE_GROUPS': ['car:2', 'truck:2', 'construction_vehicle:2', 'bus:2', 'trailer:2', 'barrier:2', 'motorcycle:2', 'bicycle:2', 'pedestrian:2', 'traffic_cone:2'], 'NUM_POINT_FEATURES': 5, 'DATABASE_WITH_FAKELIDAR': False, 'REMOVE_EXTRA_WIDTH': [0.0, 0.0, 0.0], 'LIMIT_WHOLE_SCENE': True}, {'NAME': 'random_world_flip', 'ALONG_AXIS_LIST': ['x', 'y']}, {'NAME': 'random_world_rotation', 'WORLD_ROT_ANGLE': [-0.78539816, 0.78539816]}, {'NAME': 'random_world_scaling', 'WORLD_SCALE_RANGE': [0.9, 1.1]}, {'NAME': 'random_world_translation', 'NOISE_TRANSLATE_STD': [0.5, 0.5, 0.5]}]
2023-08-09 20:08:21,915   INFO  ----------- POINT_FEATURE_ENCODING -----------
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.encoding_type: absolute_coordinates_encoding
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.used_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp']
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.POINT_FEATURE_ENCODING.src_feature_list: ['x', 'y', 'z', 'intensity', 'timestamp']
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG.DATA_PROCESSOR: [{'NAME': 'mask_points_and_boxes_outside_range', 'REMOVE_OUTSIDE_BOXES': True}, {'NAME': 'shuffle_points', 'SHUFFLE_ENABLED': {'train': True, 'test': True}}, {'NAME': 'transform_points_to_voxels', 'VOXEL_SIZE': [0.075, 0.075, 0.2], 'MAX_POINTS_PER_VOXEL': 10, 'MAX_NUMBER_OF_VOXELS': {'train': 120000, 'test': 160000}}]
2023-08-09 20:08:21,915   INFO  cfg.DATA_CONFIG._BASE_CONFIG_: cfgs/dataset_configs/nuscenes_dataset.yaml
2023-08-09 20:08:21,915   INFO  ----------- MODEL -----------
2023-08-09 20:08:21,916   INFO  cfg.MODEL.NAME: VoxelNeXt
2023-08-09 20:08:21,916   INFO  ----------- VFE -----------
2023-08-09 20:08:21,916   INFO  cfg.MODEL.VFE.NAME: MeanVFE
2023-08-09 20:08:21,916   INFO  ----------- BACKBONE_3D -----------
2023-08-09 20:08:21,916   INFO  cfg.MODEL.BACKBONE_3D.NAME: VoxelResBackBone8xVoxelNeXt
2023-08-09 20:08:21,916   INFO  ----------- DENSE_HEAD -----------
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.NAME: VoxelNeXtHead
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.CLASS_AGNOSTIC: False
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.INPUT_FEATURES: 128
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.CLASS_NAMES_EACH_HEAD: [['car'], ['truck', 'construction_vehicle'], ['bus', 'trailer'], ['barrier'], ['motorcycle', 'bicycle'], ['pedestrian', 'traffic_cone']]
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.SHARED_CONV_CHANNEL: 128
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.KERNEL_SIZE_HEAD: 1
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.USE_BIAS_BEFORE_NORM: True
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.NUM_HM_CONV: 2
2023-08-09 20:08:21,916   INFO  ----------- SEPARATE_HEAD_CFG -----------
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.SEPARATE_HEAD_CFG.HEAD_ORDER: ['center', 'center_z', 'dim', 'rot', 'vel']
2023-08-09 20:08:21,916   INFO  ----------- HEAD_DICT -----------
2023-08-09 20:08:21,916   INFO  ----------- center -----------
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.SEPARATE_HEAD_CFG.HEAD_DICT.center.out_channels: 2
2023-08-09 20:08:21,916   INFO  cfg.MODEL.DENSE_HEAD.SEPARATE_HEAD_CFG.HEAD_DICT.center.num_conv: 2
[... 44 lines cut ...]
2023-08-09 20:08:21,916   INFO  cfg.OPTIMIZATION.DECAY_STEP_LIST: [35, 45]
2023-08-09 20:08:21,916   INFO  cfg.OPTIMIZATION.LR_DECAY: 0.1
2023-08-09 20:08:21,916   INFO  cfg.OPTIMIZATION.LR_CLIP: 1e-07
2023-08-09 20:08:21,916   INFO  cfg.OPTIMIZATION.LR_WARMUP: False
2023-08-09 20:08:21,916   INFO  cfg.OPTIMIZATION.WARMUP_EPOCH: 1
2023-08-09 20:08:21,916   INFO  cfg.OPTIMIZATION.GRAD_NORM_CLIP: 10
2023-08-09 20:08:21,917   INFO  cfg.TAG: cbgs_voxel0075_voxelnext
2023-08-09 20:08:21,917   INFO  cfg.EXP_GROUP_PATH: home/wzy/3Detector/VoxelNeXt/tools/cfgs/nuscenes_models
2023-08-09 20:08:21,917   INFO  Loading NuScenes dataset
2023-08-09 20:08:21,917   INFO  Total samples for NuScenes dataset: 0
2023-08-09 20:08:25,209   INFO  ==> Loading parameters from checkpoint /home/wzy/3Detector/voxelnext_nuscenes_kernel1.pth to GPU
2023-08-09 20:08:25,240   INFO  ==> Done (loaded 542/542)
2023-08-09 20:08:25,251   INFO  *************** EPOCH 1 EVALUATION *****************
eval: 0it [00:00, ?it/s]
2023-08-09 20:08:25,420   INFO  *************** Performance of EPOCH 1 *****************
Traceback (most recent call last):
  File "test.py", line 207, in <module>
    main()
  File "test.py", line 203, in main
    eval_single_ckpt(model, test_loader, args, eval_output_dir, logger, epoch_id, dist_test=dist_test)
  File "test.py", line 65, in eval_single_ckpt
    eval_utils.eval_one_epoch(
  File "/home/wzy/3Detector/VoxelNeXt/tools/eval_utils/eval_utils.py", line 94, in eval_one_epoch
    sec_per_example = (time.time() - start_time) / len(dataloader.dataset)
### ZeroDivisionError: float division by zero

Zynade, 2023-09-17:
I attempted to replicate the results on nuScenes test dataset, and I ran into the same issue. Have you found a fix?

kbar2222, 2024-05-26:
hello, have you fix this problem plz ?

pbrandstet, 2024-06-06:
do you use the v1.0-mini dataset from nuscenes?
if so, you need to change the version.
In tools/cfgs/dataset_configs/nuscenes_dataset.yaml at line 4 you see 
VERSION: 'v1.0-trainval'
if you use mini you need to change it to
VERSION: 'v1.0-mini'

rpf1019, 2024-06-13:
> do you use the v1.0-mini dataset from nuscenes? if so, you need to change the version. In tools/cfgs/dataset_configs/nuscenes_dataset.yaml at line 4 you see VERSION: 'v1.0-trainval' if you use mini you need to change it to VERSION: 'v1.0-mini'

i tried this but it did not work, is it worked for u?

pbrandstet, 2024-06-13:
Yes, after that change it worked for me.
You could use the debugger to check if the dataset really got loaded.